wget/realtek-r8152-linux #1: Unable to build for Linux 4.20.0

hv15, 2019-01-07:
Hi there,

Tried to update to 2.10 using the DKMS package on [AUR](https://aur.archlinux.org/packages/r8152-dkms/) but this failed to compile on Linux 4.20. See attached [log file](https://github.com/wget/realtek-r8152-linux/files/2732979/make.log).

Thanks!

hcartiaux, 2019-01-07:
According to the [realtek website](https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-usb-3-0-software) where we get the driver sources, it supports kernel up to 4.15.

For the moment, I will stick with kernel 4.19 lts...

hv15, 2019-01-07:
I've come up with a fix for this, see [linux-4.20.0-add-guard-fix.patch.txt](https://github.com/wget/realtek-r8152-linux/files/2733417/linux-4.20.0-add-guard-fix.patch.txt) patch file.

Seemingly at some point between 4.19 and 4.20 development, the ethtool interface was changed and no longer supported certain callbacks. Realtek already implemented the correct (new) callbacks, but was still using the old interface. The patch just adds a guard around the affected area.

Seems to be working fine on my system.

hcartiaux, 2019-01-08:
I've tested your patch and it works on my system too.
I've integrated your patch on AUR in the PKGBUILD.

Thank you for your help !

closed 2019-01-08 by hcartiaux

wget, 2019-01-09:
@hcartiaux @hv15 Super work guys :)
Today, I was rebooting my computer and was that DKMS issue. I connected to my AUR and my github, I saw you had already taken care of it. Thanks!