wget/realtek-r8152-linux #2: Do not work in usb 3.0 at Dell Inspiron

JuniorPolegato, 2019-11-14:
Hi! My device do not work in 3.0, so I get up to 350 Mbps max. How can I force to use 3.0? Thanks.

# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 1: Dev 4, If 0, Class=Vendor Specific Class, Driver=r8152, 480M
    |__ Port 6: Dev 2, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 6: Dev 2, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 10: Dev 3, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 10: Dev 3, If 1, Class=Wireless, Driver=btusb, 12M

# dmesg
[ 2787.398035] usb 1-1: new high-speed USB device number 5 using xhci_hcd
[ 2787.547490] usb 1-1: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
[ 2787.547496] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[ 2787.547500] usb 1-1: Product: USB 10/100/1000 LAN
[ 2787.547503] usb 1-1: Manufacturer: Realtek
[ 2787.547505] usb 1-1: SerialNumber: 000001
[ 2787.568368] usbcore: registered new interface driver r8152
[ 2787.694538] usb 1-1: reset high-speed USB device number 5 using xhci_hcd
[ 2787.900264] r8152 1-1:1.0 eth0: v2.12.0 (2019/04/29)
[ 2787.900272] r8152 1-1:1.0 eth0: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 2787.920253] r8152 1-1:1.0 enx00e04c6822c2: renamed from eth0

# lspci -nn | grep USB
00:14.0 USB controller [0c03]: Intel Corporation Cannon Point-LP USB 3.1 xHCI Controller [8086:9ded] (rev 30)

wget, 2019-11-14:
@JuniorPolegato Could you report to me the details returned by `ethtool <interface name>`?

JuniorPolegato, 2019-11-20:
Hi!

I suspect some inoperatibility about Intel USB 3.x kernel module with this realtek-r8152-linux. Is it possible force the kernel to recognize this device like a USB 3.x device? Thanks.

root@notemovel:~# inxi -xxxzF --usb
System:    Host: jumovel Kernel: 5.2.0-3-amd64 x86_64 bits: 64 compiler: gcc v: 8.3.0 Console: tty 4 dm: GDM3 3.34.1 
           Distro: Debian GNU/Linux bullseye/sid 
Machine:   Type: Laptop System: Dell product: Inspiron 3480 v: N/A serial: <filter> Chassis: type: 10 serial: <filter> 
           Mobo: Dell model: 0TDN17 v: A01 serial: <filter> UEFI: Dell v: 1.6.0 date: 07/11/2019 
Battery:   ID-1: BAT0 charge: 42.0 Wh condition: 42.0/42.0 Wh (100%) volts: 13.0/11.4 model: BYD DELL FW8KR98 type: Li-ion 
           serial: <filter> status: Full 
CPU:       Topology: Quad Core model: Intel Core i5-8265U bits: 64 type: MT MCP arch: Kaby Lake rev: C L1 cache: 256 KiB 
           L2 cache: 6144 KiB L3 cache: 6144 KiB 
           flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 28800 
           Speed: 818 MHz min/max: 400/3900 MHz Core speeds (MHz): 1: 818 2: 846 3: 893 4: 805 5: 873 6: 835 7: 837 8: 800 
Graphics:  Device-1: Intel UHD Graphics 620 vendor: Dell driver: i915 v: kernel bus ID: 00:02.0 chip ID: 8086:3ea0 
           Display: server: X.org 1.20.4 driver: i915 compositor: gnome-shell tty: 142x34 
           Message: Advanced graphics data unavailable in console for root. 
Audio:     Device-1: Intel Cannon Point-LP High Definition Audio vendor: Dell driver: snd_hda_intel v: kernel bus ID: 00:1f.3 
           chip ID: 8086:9dc8 
           Sound Server: ALSA v: k5.2.0-3-amd64 
Network:   Device-1: Intel Cannon Point-LP CNVi [Wireless-AC] driver: iwlwifi v: kernel port: 4000 bus ID: 00:14.3 
           chip ID: 8086:9df0 
           IF: wlo1 state: up mac: <filter> 
           Device-2: Realtek RTL810xE PCI Express Fast Ethernet vendor: Dell driver: r8169 v: kernel port: 3000 
           bus ID: 01:00.0 chip ID: 10ec:8136 
           IF: enp1s0 state: down mac: <filter> 
           Device-3: Realtek RTL8153 Gigabit Ethernet Adapter type: USB driver: r8152 bus ID: 1-2:30 chip ID: 0bda:8153 
           serial: <filter> 
           IF: enx00e04c68264b state: up speed: 1000 Mbps duplex: full mac: <filter> 
Drives:    Local Storage: total: 1.02 TiB used: 622.65 GiB (59.4%) 
           ID-1: /dev/sda vendor: Western Digital model: WD10SPZX-75Z10T2 size: 931.51 GiB speed: 6.0 Gb/s rotation: 5400 rpm 
           serial: <filter> rev: 1A03 temp: 40 C scheme: GPT 
           ID-2: /dev/sdb type: USB vendor: Generic model: SD MMC MS PRO size: 116.43 GiB serial: <filter> rev: 1.00 
Partition: ID-1: / size: 4.86 GiB used: 90.3 MiB (1.8%) fs: ext4 dev: /dev/sda2 
           ID-2: /home size: 752.99 GiB used: 590.88 GiB (78.5%) fs: ext4 dev: /dev/sda7 
           ID-3: /tmp size: 48.97 GiB used: 7.78 GiB (15.9%) fs: ext4 dev: /dev/sda6 
           ID-4: /usr size: 48.97 GiB used: 21.13 GiB (43.2%) fs: ext4 dev: /dev/sda3 
           ID-5: /var size: 48.97 GiB used: 2.41 GiB (4.9%) fs: ext4 dev: /dev/sda4 
           ID-6: swap-1 size: 10.00 GiB used: 358.2 MiB (3.5%) fs: swap dev: /dev/sda5 
USB:       Hub: 1-0:1 info: Full speed (or root) Hub ports: 12 rev: 2.0 speed: 480 Mb/s chip ID: 1d6b:0002 
           Device-1: 1-2:30 info: Realtek RTL8153 Gigabit Ethernet Adapter type: Network driver: r8152 interfaces: 1 rev: 2.1 
           speed: 480 Mb/s chip ID: 0bda:8153 serial: <filter> 
           Device-2: 1-3:2 info: USB OPTICAL MOUSE type: Mouse driver: hid-generic,usbhid interfaces: 1 rev: 1.1 
           speed: 1.5 Mb/s chip ID: 275d:0ba6 
           Device-3: 1-6:3 info: Realtek Integrated_Webcam_HD type: Video driver: uvcvideo interfaces: 2 rev: 2.0 
           speed: 480 Mb/s chip ID: 0bda:5520 serial: <filter> 
           Device-4: 1-7:16 info: Realtek USB2.0-CRW type: Mass Storage driver: ums-realtek interfaces: 1 rev: 2.0 
           speed: 480 Mb/s chip ID: 0bda:0177 serial: <filter> 
           Device-5: 1-10:4 info: Intel type: Bluetooth driver: btusb interfaces: 2 rev: 2.0 speed: 12 Mb/s chip ID: 8087:0aaa 
           Hub: 2-0:1 info: Full speed (or root) Hub ports: 6 rev: 3.0 speed: 5 Gb/s chip ID: 1d6b:0003 
Sensors:   System Temperatures: cpu: 54.0 C mobo: N/A 
           Fan Speeds (RPM): cpu: 0 
Info:      Processes: 316 Uptime: 1d 6h 05m Memory: 15.59 GiB used: 4.75 GiB (30.5%) Init: systemd v: 243 runlevel: 5 
           Compilers: gcc: 9.2.1 alt: 8/9 Shell: bash v: 5.0.11 running in: tty 4 inxi: 3.0.36 

root@notemovel:~# ethtool enx00e04c68264b
Settings for enx00e04c68264b:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Half 1000baseT/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
[... 13 lines cut ...]
	Supports Wake-on: pumbg
	Wake-on: g
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes

root@notemovel:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
[... cut ...]
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/12p, 480M
    |__ Port 2: Dev 30, If 0, Class=Vendor Specific Class, Driver=r8152, 480M
    |__ Port 3: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 6: Dev 3, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 6: Dev 3, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 16, If 0, Class=Mass Storage, Driver=ums-realtek, 480M
    |__ Port 10: Dev 4, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 10: Dev 4, If 1, Class=Wireless, Driver=btusb, 12M

wget, 2019-11-29:
@JuniorPolegato I have debugged a bit from my side, but couldn't find the reason why this isn't working. Would you mind asking the question on the mailing list arch-general and put me in copy [(my email address is in the PKGBUILD file)](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=r8152-dkms)?

wget, 2019-11-29:
@JuniorPolegato Btw, I realized the [udev rules](https://github.com/wget/realtek-r8152-linux/blob/master/50-usb-realtek-net.rules) weren't applied AFAIK. I'll publish a dkms patch for this.

wget, 2020-02-05:
Ok. I have just realized that this bug report was concerning a problem from upstream with a specific USB device rather than from the Arch Linux package I maintain as I cannot reproduce with my stack (Arch Linux + dongles).

Also note that even if a dongle is using this driver, this doesn't mean this will work completely. I have a no name RTL8152 bought from Amazon and that one is not working nicely when being under load. I replaced it two times now and it's still the same issue.

Also, there is now a new UEFI update [for your computer](https://www.dell.com/support/home/be/fr/bebsdt1/product-support/product/inspiron-14-3480-laptop/drivers) (update v 1.7.0). I encourage you to upgrade. Have noticed weird things with USB dongles in the past, problem fading away and disappearing with UEFI updates.

closed 2020-02-05 by wget