wget/realtek-r8152-linux #3: Random hangs with Tx status -71

phpbg, 2020-01-29:
Hi,
I sometimes experience random network hangs with those messages in dmesg:
```
kernel: r8152 2-1.4:1.0 enx503eaad13fed: Tx status -71
```
I have to unplug / replug network adapter to recover.

Any idea why?

phpbg, 2020-01-29:
NB: last time it happened: `Linux version 5.3.0-26-generic (buildd@lgw01-amd64-039) (gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1))`

wget, 2020-02-05:
I have just realized that this bug report was concerning a problem from upstream with a specific USB device rather than from the Arch Linux package I maintain as I cannot reproduce with my stack (Arch Linux + dongles).

Also note that even if a dongle is using this driver, this doesn't mean this will work completely. I have a no name RTL8152 bought from Amazon and that one is not working nicely when being under load. I replaced it two times now and it's still the same issue.

Sometimes, I have been able to reproduce the disconnection issue you mention, but this only happened with the cheap RTL8152 based dongle I have.

This repo is meant for sourcing easily the RTL8152 package I maintain for Arch Linux, not to report upstream bugs. Closing since this seems to be a device specific problem and in all cases an upstream problem.

closed 2020-02-05 by wget