wget/realtek-r8152-linux #6: Install instructions for noobs?

moose09876, 2020-07-27:
Not sure how to resolve the issue I'm seeing after doing a sudo make install:

root@r:/mnt/md0/realtek-r8152-linux-master# sudo make install
rmmod r8152
make -C /lib/modules/5.4.0-42-generic/build M=/mnt/md0/realtek-r8152-linux-master INSTALL_MOD_DIR=kernel/drivers/net/usb modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-42-generic'
  INSTALL /mnt/md0/realtek-r8152-linux-master/r8152.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-42-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-42-generic'
modprobe r8152
root@r:/mnt/md0/realtek-r8152-linux-master#

I'm guessing it's something I'm not doing correctly. Any help would be appreciated.

wget, 2020-07-28:
@moose09876 Hi there,

This is not an error per se.

What you are seeing here is that make install tries to sign your kernel module. Depending on your distribution, this steps may be required (Ubuntu for example, on Arch Linux the signing is not enforced yet).

If I understand your logs correctly, with the line `modprobe r8152` it seems like the module is still properly loaded. So even if mentioning the word `error` it may just be a warning after all.

I encourage you to provide more pieces of info (distribution being used) and to check the way to build kernel modules for your specific distribution.

Like mentioned in the description of this repo:
> Meant to be used in distributions only - For upstream bugs, please report them to your distribution maintainer or to Realtek. 

People using this module need no know what they are doing :) This repo is basically a mirror to avoid accesc issues when automating this package. The realtek website is needed a pain to deal with in terms of session.

closed 2020-07-28 by wget

moose09876, 2020-07-28:
Thanks for the response. I'll never know what I'm doing in linux, lol, just trying to get this adapter working properly. I'll reach out to the manufacturer again.