wget/realtek-r8152-linux #7: Tested with 5.8.1 r8156

autarctechelias, 2020-08-15:
Not really an issue per se but it works perfectly fine under 5.8.1-generic using a usb-c to r8156.
uname -a: Linux elias-ThinkPad-X250 5.8.1-050801-generic #202008111432 SMP Tue Aug 11 14:34:42 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
lsusb: Bus 003 Device 007: ID 0bda:8156 Realtek Semiconductor Corp. 

Hope this helps.

wget, 2020-08-19:
@autarctechelias Thanks for letting me notice this.

The udev rules is already making sure to use this driver for 8156 based adapters though.
https://github.com/wget/realtek-r8152-linux/blob/master/50-usb-realtek-net.rules#L11

Does your question imply that this driver is not being used automatically when your insert your Realtek 8156 adapter? If that's the case, is there any chance this driver enters in conflict with an existing driver on your distribution?

hcartiaux, 2022-09-14:
No activity in 2 years. Let's close.

closed 2022-09-14 by hcartiaux