wget/realtek-r8152-linux #8: make install error Ubuntu 20

FilipLaurentiu, 2020-11-03:
```
filip@filip-pc:~/Desktop/realtek-r8152-linux$ sudo make install
\rmmod r8152
make -C /lib/modules/5.4.0-52-generic/build M=/home/filip/Desktop/realtek-r8152-linux INSTALL_MOD_DIR=kernel/drivers/net/usb modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-52-generic'
  INSTALL /home/filip/Desktop/realtek-r8152-linux/r8152.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory

```

FilipLaurentiu, 2020-11-03:
Can you help me with this error ?

wget, 2020-11-03:
@FilipLaurentiu Hi Filip :) 

This is not an issue per se. As you see the compilation process just complains it cannot find a proper private key to sign the kernel module. Indeed, Ubuntu has stricter policies wrt. kernel modules and requires to have them signed, especially on UEFI with Secure Boot enabled.

If you need to have the module signed, just follow [this tutorial](https://superuser.com/a/1513506).

As for where you need to put the private key you'll need to enroll, just check the relative path `certs/signing_key.pem`.

Hope this gets you on track :) Feel free to reach out if you have any issue :)

FilipLaurentiu, 2020-11-03:
Hmm, thanks for help.

I follow this tutorial instead [oficial](https://ubuntu.com/blog/how-to-sign-things-for-secure-boot) it was much simpler but not much success

```
filip@filip-pc:~/Downloads/r8152-2.14.0$ sudo make install
rmmod r8152
make -C /lib/modules/5.4.0-52-generic/build M=/home/filip/Downloads/r8152-2.14.0 INSTALL_MOD_DIR=kernel/drivers/net/usb modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-52-generic'
  INSTALL /home/filip/Downloads/r8152-2.14.0/r8152.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-52-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-52-generic'
modprobe r8152
filip@filip-pc:~/Downloads/r8152-2.14.0$ hexdump -Cv r8152.ko | tail -n 5
000443d0  4a 9f 96 72 e8 c3 1d 21  9d 69 47 9c d1 5a f0 a7  |J..r...!.iG..Z..|
000443e0  00 00 02 00 00 00 00 00  00 00 01 f0 7e 4d 6f 64  |............~Mod|
000443f0  75 6c 65 20 73 69 67 6e  61 74 75 72 65 20 61 70  |ule signature ap|
00044400  70 65 6e 64 65 64 7e 0a                           |pended~.|
00044408
filip@filip-pc:~/Downloads/r8152-2.14.0$ 
```

FilipLaurentiu, 2020-11-03:
> @FilipLaurentiu Hi Filip :)
>
> This is not an issue per se. As you see the compilation process just complains it cannot find a proper private key to sign the kernel module. Indeed, Ubuntu has stricter [...]

Thank you for your help !

I generate a key and load it base on [official](https://ubuntu.com/blog/how-to-sign-things-for-secure-boot) documentation

then I copy the private key in the driver folder but it not help
```
filip@filip-pc:~/Downloads/r8152.53.56-2.14.0/r8152-2.14.0$ cat certs/signing_key.pem.priv 
-----BEGIN PRIVATE KEY-----
MIIEvgIBADANBgkqhkiG9w0BAQEFAASCBKgwggSkAgEAAoIBAQCn5SfsYtH1wvNL
......
MXcgCLtn09DKZ2MKUjpXo7c0
-----END PRIVATE KEY-----
filip@filip-pc:~/Downloads/r8152.53.56-2.14.0/r8152-2.14.0$ sudo make install
make -C /lib/modules/5.4.0-52-generic/build M=/home/filip/Downloads/r8152.53.56-2.14.0/r8152-2.14.0 INSTALL_MOD_DIR=kernel/drivers/net/usb modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-52-generic'
  INSTALL /home/filip/Downloads/r8152.53.56-2.14.0/r8152-2.14.0/r8152.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-52-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-52-generic'
modprobe r8152
modprobe: ERROR: could not insert 'r8152': Operation not permitted
make: *** [Makefile:42: install] Error 1
```

FilipLaurentiu, 2020-11-03:
I also try to manually sign the module

```
filip@filip-pc:~/Downloads/r8152.53.56-2.14.0/r8152-2.14.0$ hexdump -Cv r8152.ko | tail -n 5
000441b0  3a 15 4f 11 34 55 cc e4  8d 58 47 ca aa b5 ca dd  |:.O.4U...XG.....|
000441c0  c2 13 9c 8d 80 cb 0a c3  00 00 02 00 00 00 00 00  |................|
000441d0  00 00 01 f0 7e 4d 6f 64  75 6c 65 20 73 69 67 6e  |....~Module sign|
000441e0  61 74 75 72 65 20 61 70  70 65 6e 64 65 64 7e 0a  |ature appended~.|
000441f0
filip@filip-pc:~/Downloads/r8152.53.56-2.14.0/r8152-2.14.0$ sudo make install
make -C /lib/modules/5.4.0-52-generic/build M=/home/filip/Downloads/r8152.53.56-2.14.0/r8152-2.14.0 INSTALL_MOD_DIR=kernel/drivers/net/usb modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-52-generic'
  INSTALL /home/filip/Downloads/r8152.53.56-2.14.0/r8152-2.14.0/r8152.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.4.0-52-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-52-generic'
modprobe r8152

```

scooter-dangle, 2021-01-17:
I'm totally out of my depth on this, but I get an identical error (apart from my kernel version being `5.4.0-62-generic`). A number of similar driver install issues seem to be resolved by using `dkms` (e.g., patjak/bcwc_pcie#176).

My rough stab at a `dkms.conf` file (so that I could follow the steps at https://stackoverflow.com/a/46438765/808850) didn't work, but it looked like this:
```
MAKE="make modules"
CLEAN="make clean"
BUILT_MODULE_NAME=r8152
BUILT_MODULE_LOCATION=./
DEST_MODULE_LOCATION=/kernel/drivers/net/usb/
PACKAGE_NAME=r8152
PACKAGE_VERSION=2.14.0
REMAKE_INITRD=yes
```

Is this completely the wrong track? Maybe the proper content of `dkms.conf` is more obvious to someone else.

mouse3dash7, 2021-03-15:
Try:

1. Install DKMS
> sudo apt install dkms
2. Extract source code:
> sudo tar -xvf r8152.53.56-2.14.0.tar -C /usr/src/
3. Create dkms.conf
> sudo nano /usr/src/r8152-2.14.0/dkms.conf
======================================
PACKAGE_NAME="r8152"
PACKAGE_VERSION="2.14.0"
BUILT_MODULE_NAME[0]="$PACKAGE_NAME"
DEST_MODULE_LOCATION[0]="/updates/dkms"
AUTOINSTALL="YES"
REMAKE_INITRD="YES"
CLEAN="rm src/@PKGNAME@.ko src/*.o || true"
======================================
4. Install driver:
> sudo dkms add -m r8152 -v 2.14.0
> sudo dkms build -m r8152 -v 2.14.0
> sudo dkms install -m r8152 -v 2.14.0
> sudo depmod -a
> sudo modprobe r8152

hcartiaux, 2022-09-14:
The issue is not related to the module sources.

I close.

closed 2022-09-14 by hcartiaux

dolikemedo, 2023-03-09:
Hi!
Sorry, I got my hopes up, only last line returns error:

modprobe: ERROR: could not insert 'r8152': Exec format error

Thank you if someone can help.