wget/realtek-r8152-linux #9: How can i add this driver to kernel while cross compiling kernel for raspberry pi zero ?

bogaskilop, 2020-11-17:
I'm about to start compiling new kernel for rpi zero   and i wanted to add this driver to my kernel but can you point me how can i add this driver  and compile new kernel with this new driver  2.14 ?   as i occurs some problems even with  kernel 5.4.75  for rpi   and that driver is  1.10.x version  and  rtl8152 cards not always  working  most of the time enumerate problems .

Thx M

bogaskilop, 2020-11-18:
ok i found simpler and quicker sollution to get headers   witout compiling new kernel :
[https://github.com/RPi-Distro/rpi-source](https://github.com/RPi-Distro/rpi-source)

so driver is compiled now test time

bogaskilop, 2020-11-19:
but with new drivers  2.14 and 5.4.77 kernel rpi zero not even ones  after  6 boots  turn on network card :( so i need to find different lan adapters  :(

hcartiaux, 2022-09-14:
Out of topic.