wget/realtek-r8152-linux #10: Difference between latest 2.14 version and the one in the stock kernel?

The9Cat, 2020-11-18:
This is working well for me thanks.  It seems to have solved a timeout problem that I was having with a usb-c ethernet adapter.

But I'm curious: what is the difference between this driver from Realtek and the one in the stock kernel?  How can one figure that out?

ulli-kroll, 2020-11-25:
s/stock/vanilla/

-support for NBaseT with 2.5Gbit/s
-as for 2.14 support for some IIRC DELL hardware with 1Gbit/s
..  this is somewhere is the code

The9Cat, 2020-11-25:
Thanks.

I've noticed that the r8153 usb works with 2.14 but times out like crazy
with the default kernel module.  Tried this with several manufacturers'
implementations and it is the same.  Wondering if this is difference in
firmware blobs or something in the module source . . .

I suppose it could be my host hardware, but that seems less likely to me.

ulli-kroll, 2020-11-28:
You mean r8153 device ?
IIRC with gibabit ethernet.

It maybe the driver itselt or part of the firmware, which is builtin this *very* driver.
upstream/vanilla driver has some extra file for this. With extra parsing schematics 
inside the driver protected by sha256 (IIRC).
It may also the host controller, because on one of my embedded devices I get lots of protocol errors (errno -71), maybe due the XHCI implementation.

closed 2022-12-09 by hcartiaux