wget/realtek-r8152-linux #16: New 5.17 kernel causes issues with compile

preese, 2022-03-20:
Expecting this will be a problem when kernel 5.17 goes live.  This is from kernel 5.17-rc8.

Seems like a similar issue to the recent get/set_coalesce one.  I'm not a programmer so have little idea of what the issue is but have VERY MUCH enjoyed using this driver over the past months since I found this gem.

Thanks for taking a look.
Phil

```
/home/preese/realtek-r8152-linux-2.15.20211119/r8152.c: At top level:
/home/preese/realtek-r8152-linux-2.15.20211119/r8152.c:18560:26: error: initialization of ‘void (*)(struct net_device *, struct ethtool_ringparam *, struct kernel_ethtool_ringparam *, struct netlink_ext_ack *)’ from incompatible pointer type ‘void (*)(struct net_device *, struct ethtool_ringparam *)’ [-Werror=incompatible-pointer-types]
18560 |         .get_ringparam = rtl8152_get_ringparam,
      |                          ^~~~~~~~~~~~~~~~~~~~~
/home/preese/realtek-r8152-linux-2.15.20211119/r8152.c:18560:26: note: (near initialization for ‘ops.get_ringparam’)
/home/preese/realtek-r8152-linux-2.15.20211119/r8152.c:18561:26: error: initialization of ‘int (*)(struct net_device *, struct ethtool_ringparam *, struct kernel_ethtool_ringparam *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_ringparam *)’ [-Werror=incompatible-pointer-types]
18561 |         .set_ringparam = rtl8152_set_ringparam,
      |                          ^~~~~~~~~~~~~~~~~~~~~
/home/preese/realtek-r8152-linux-2.15.20211119/r8152.c:18561:26: note: (near initialization for ‘ops.set_ringparam’)
```
There was a warning earlier in the MAKE but I don't think it is a show stopper:

```
/home/preese/realtek-r8152-linux-2.15.20211119/r8152.c: In function ‘rtl8152_set_mac_address’:
/home/preese/realtek-r8152-linux-2.15.20211119/r8152.c:1435:22: warning: passing argument 1 of ‘memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1435 |         memcpy(netdev->dev_addr, addr->sa_data, netdev->addr_len);
      |                ~~~~~~^~~~~~~~~~
In file included from ./include/linux/string.h:253,
                 from ./include/linux/bitmap.h:11,
                 from ./include/linux/cpumask.h:12,
... 
                 from /home/preese/realtek-r8152-linux-2.15.20211119/r8152.c:13:
./include/linux/fortify-string.h:212:37: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  212 | __FORTIFY_INLINE void *memcpy(void *p, const void *q, __kernel_size_t size)
      |                               ~~~~~~^
/home/preese/realtek-r8152-linux-2.15.20211119/r8152.c: In function ‘set_ethernet_addr’:
/home/preese/realtek-r8152-linux-2.15.20211119/r8152.c:1561:36: warning: passing argument 1 of ‘ether_addr_copy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1561 |                 ether_addr_copy(dev->dev_addr, sa.sa_data);
      |                                 ~~~^~~~~~~~~~
In file included from /home/preese/realtek-r8152-linux-2.15.20211119/r8152.c:17:
./include/linux/etherdevice.h:295:40: note: expected ‘u8 *’ {aka ‘unsigned char *’} but argument is of type ‘const unsigned char *’
  295 | static inline void ether_addr_copy(u8 *dst, const u8 *src)
      |                                    ~~~~^~~
```

preese, 2022-03-23:
Update on the above.  Using the file "50-usb-realtek-net.rules" solves connectivity issues once I did what was needed to swap the cdc_mbin driver for the OS's stock r8152 driver.  This got things working and at the 2.5g speed but uses the much older v1.12.12 version of the driver.  I'm back to working but the newer driver would be a 'nice to have'.

schristiany, 2022-03-26:
the error message tells, that a warning got converted to an error. I  just removed -Werror=incompatible-pointer-types out of Makefile. 

The resulting kernel works, but I get an error at boot (XXXed the Mac):
`
[Thu Mar 24 12:32:10 2022] r8152 2-2:1.0 enxXXX: Current addr: XX XX 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[Thu Mar 24 12:32:10 2022] r8152 2-2:1.0 enxXXX: Expected addr: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[Thu Mar 24 12:32:10 2022] ------------[ cut here ]------------
[Thu Mar 24 12:32:10 2022] netdevice: enxXXX: Incorrect netdev->dev_addr
`
despite this, network is working properly at 2.5G. Manually switching the connection  speed is working too.

preese, 2022-03-26:
Thanks[schristiany](https://github.com/schristiany)!  I'm not positive of your suggestion but it got me thinking and tinkering.  I compiled the original RealTek code on a new Fedora 36 pre-beta OS, it uses kernel version 5.17.0-300.fc36.x86_64.  I saw the coalesce errors and then the ringparm errors.  The errors looked surprisingly similar.  It seems some of the newer libraries have gone for fewer elements in their STRUCT format.  So I applied similar patches as was done for the coalesce fix.  To my happy surprise, they seemed to work just fine, no obvious errors, though I don't really know where to look.  I've used this with the new kernel and I've added it to a F35 system's dkms system and then updated to F36 with success!

I'm still of the opinion that your life is easier if you use a RealTek based USB3 dongle, with a lsusb ID of 0BD:815(2 3 5
 6).  I did get the ASUS dongle to work (ID of 0b05:1976) but the bits need to be added to the r8152.c source file and to the 50-usb-realtek-net.rules file.

So here are the patches:  (I added them when 'void rtl8152_get_ringparam' and  'int rtl8152_set_ringparam' were first found in the source.)

```
#if LINUX_VERSION_CODE < KERNEL_VERSION(5,15,0)
static void rtl8152_get_ringparam(struct net_device *netdev,
                  struct ethtool_ringparam *ring)
#else
static void rtl8152_get_ringparam(struct net_device *netdev,
                  struct ethtool_ringparam *ring,
                  struct kernel_ethtool_ringparam *kernel_ring,
                  struct netlink_ext_ack *extack)
#endif

and 

#if LINUX_VERSION_CODE < KERNEL_VERSION(5,15,0)
static int rtl8152_set_ringparam(struct net_device *netdev,
                 struct ethtool_ringparam *ring)
#else
static int rtl8152_set_ringparam(struct net_device *netdev,
                 struct ethtool_ringparam *ring,
                 struct kernel_ethtool_ringparam *kernel_ring,
                 struct netlink_ext_ack *extack)
#endif
```
As is evident from my posts, I'm new at this stuff.  I'm not sure how to use Git enough to push my patched version of r8152.c back to the Main, sorry. If someone does know and can review the sanity for the patches please do the GIT work, and you'll have my thanks!

schristiany, 2022-03-29:
@preese my ugly hack was for testing only. 
Thanks for the patch, it  works for me. However, I still get the same calltrace as before.

hcartiaux, 2022-04-01:
@preese I think you're right, this is in the diff from 5.16 to 5.17: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/net/usb/r8152.c?id=7462494408cd3de8b0bc1e79670bf213288501d0

I commit a fix and I close this bug.

commit 54b8e299 (2022-04-01): Merge pull request #17 from hcartiaux/master

Import changes from linux 5.17 for compatibility (closes #16)
closed 2022-04-01 by hcartiaux