wget/realtek-r8152-linux #22: v2.16.1 on kernel 5.19.1 netif_set_gso....???netif_set_tso???

ohquait, 2022-08-12:
DKMS make.log for r8152-2.16.1-1pclos2022 for kernel 5.19.1-pclos1 (x86_64)
Fri Aug 12 08:57:40 AM UTC 2022
make	 -C /lib/modules/5.19.1-pclos1/build M=/var/lib/dkms/r8152/2.16.1-1pclos2022/build modules
make[1]: Entering directory '/usr/src/kernel-devel-5.19.1-pclos1'
  CC [M]  /var/lib/dkms/r8152/2.16.1-1pclos2022/build/r8152.o
/var/lib/dkms/r8152/2.16.1-1pclos2022/build/r8152.c: In function ‘sg_en_store’:
/var/lib/dkms/r8152/2.16.1-1pclos2022/build/r8152.c:20359:9: error: implicit declaration of function ‘netif_set_gso_max_size’; did you mean ‘netif_set_tso_max_size’? [-Werror=implicit-function-declaration]
20359 |         netif_set_gso_max_size(netdev, tso_size);
      |         ^~~~~~~~~~~~~~~~~~~~~~
      |         netif_set_tso_max_size
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/r8152/2.16.1-1pclos2022/build/r8152.o] Error 1
make[1]: *** [Makefile:1836: /var/lib/dkms/r8152/2.16.1-1pclos2022/build] Error 2
make[1]: Leaving directory '/usr/src/kernel-devel-5.19.1-pclos1'
make: *** [Makefile:30: modules] Error 2

ohquait, 2022-08-12:
may be new version 2.16.3 can solve

menzo2003, 2022-08-14:
I get the same error on 2.16.3

Gzxhwq, 2022-08-16:
https://bugs.launchpad.net/ubuntu/+source/r8168/+bug/1982000

ohquait, 2022-08-17:
this working on r8152-2.16.1, i used the r8152.c file from to build a dkms-r8152-2.16.3 package, the source of real 2.16.3 is not here ?
```
# apt-get install dkms-r8152
Reading Package Lists... Done
Building Dependency Tree... Done
The following NEW packages will be installed:
   dkms-r8152 (2.16.3-2pclos2022)
0 upgraded, 1 newly installed, 0 removed and 0 not upgraded.
Need to get 80.0kB of archives.
After unpacking 653kB of additional disk space will be used.
Get:1 http://qua-it.org quait/2022/x86_64 dkms-r8152 2.16.3-2pclos2022 [80.0kB]
Fetched 80.0kB in 0s (476kB/s)  
Committing changes...
Preparing                                ############################## [100%]
Updating / installing
  dkms-r8152-2.16.3-2pclos2022.x86_64    ############################## [100%]

Creating symlink /var/lib/dkms/r8152/2.16.3-2pclos2022/source ->
                 /usr/src/r8152-2.16.3-2pclos2022

DKMS: add Completed.

Preparing kernel 5.19.1-pclos2 for module build:
(This is not compiling a kernel, just preparing kernel symbols)
Storing current .config to be restored when complete
Running Generic preparation routine
make clean.....
using /proc/config.gz
make oldconfig....
make prepare....

Building module:
cleaning build area....
export IGNORE_CC_MISMATCH=1; 'make' KSRC=/lib/modules/5.19.1-pclos2/build EXTRA_CFLAGS='-DCONFIG_R8152_NAPI -DCONFIG_R8152_VLAN -DRTL8152_S5_WOL' SKIP_STACK_VALIDATION=1 all.....
cleaning build area....
cleaning kernel tree (make clean).....

DKMS: build Completed.

r8152.ko.xz:
 - Installation
   - Installing to /lib/modules/5.19.1-pclos2/kernel/3rdparty/r8152-53/

depmod.....

DKMS: install Completed.
Reloading network configuration ... 
Done.

```
**with the patch below, the build works, but i can not knowing if this will working in real, i don't have a adapter like for r8152....**

```
--- a/r8152.c
+++ b/r8152.c
@@ -20355,9 +20355,9 @@
 		return -EINVAL;
 	}
 
-#if LINUX_VERSION_CODE >= KERNEL_VERSION(2,6,26)
-	netif_set_gso_max_size(netdev, tso_size);
-#endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(2,6,26) */
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5,19,0)
+	netif_set_tso_max_size(netdev, tso_size);
+#endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(5,19,0) */
 
 	return count;
 }
@@ -20517,12 +20517,9 @@
 	rtl_get_mapt_ver(tp);
 
 	netdev->ethtool_ops = &ops;
-#if LINUX_VERSION_CODE >= KERNEL_VERSION(2,6,26)
-	if (!tp->sg_use)
-		netif_set_gso_max_size(netdev, RTL_LIMITED_TSO_SIZE);
-#else
-	netdev->features &= ~(NETIF_F_TSO | NETIF_F_TSO6);
-#endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(2,6,26) */
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5,19,0)
+	netif_set_tso_max_size(netdev, RTL_LIMITED_TSO_SIZE);
+#endif /* LINUX_VERSION_CODE >= KERNEL_VERSION(5,19,0) */
 
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(4,10,0)
 	/* MTU range: 68 - 1500 or 9194 */

```

menzo2003, 2022-08-17:
I tested it with my Asus USB-C-2500 that uses this driver, and your patch works on kernel 5.19.1-arch2-1 with version 2.16.3

Thanks!

ohquait, 2022-08-17:
can you give me working link to source of 2.16.3 ?

menzo2003, 2022-08-17:
https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-usb-3-0-software

From the realtek website.

ohquait, 2022-08-17:
ok, thank you, but capcha-code not loving me.
from[https://github.com/MattFromGer/r8152.53.56-linux](url)
i have it, and if who will, here are also:
[https://www.qua-it.org/XFCE/shots/share-with-pclos/pclos/r8152-2.16.3.tar.xz](url)

menzo2003, 2022-08-17:
If it gives you the sum it wants you to answer it not retype it, it is very stupid.

ohquait, 2022-08-18:
sorry that's the age

Barbyie, 2022-09-04:
How can I apply this patch ? I'm getting the same error.

hcartiaux, 2022-09-14:
I updated the repository with the latest version and a fix for Linux 5.19.

I close this issue.

closed 2022-09-14 by hcartiaux