wget/realtek-r8152-linux #24: 6.1 kernel API changes

ervinpopescu, 2022-11-07:
`netif_napi_add` no longer takes napi poll weight as parameter, 6.1 defines it as macro: https://elixir.bootlin.com/linux/v6.1-rc1/source/include/linux/netdevice.h#L2547

https://elixir.bootlin.com/linux/v6.0.7/source/include/linux/netdevice.h#L2560
https://elixir.bootlin.com/linux/v6.1-rc1/source/include/linux/netdevice.h#L2562

jayofdoom, 2022-11-15:
Hey; thanks for this. I'm looking to take this patch downstream in Gentoo; do you have an email you'd like me to credit? e.g. https://github.com/gentoo/gentoo/blob/master/net-misc/r8152/files/r8152-2.16.3-kernel-5.19-fix.patch 

Alternatively; if you have a fork with this commit it'd be trivial for me to turn that into a git patch. Thanks!

ervinpopescu, 2022-11-15:
Hello there!

Yeah, I guess `ervin.popescu10@gmail.com` will do fine. And I'll try pushing a commit to a fork ASAP.

hcartiaux, 2022-12-09:
Thanks for this patch. 

I have imported your commit in master. Do not hesitate to create a PR next time ;)

closed 2022-12-09 by hcartiaux