wget/realtek-r8152-linux #26: What exactly is "center tap short"?

HerMajestyDrMona, 2023-01-09:
Hi. I'm trying to fix performance issue with my new USB3.0 to 2.5Gb Ethernet adapter.
The READ ME page describes a method to compile with:
`make CONFIG_CTAP_SHORT=OFF modules`

I wonder what this option changes? Should it theoretically improve performance?

jayofdoom, 2023-03-16:
Hey, just letting you know I saw this. I have no idea what it means. It's referenced in the upstream code from Realtek.

joecool1029, 2023-08-22:
In short: Don't use it if your adapter is negotiating as expected. 

More background: May have to email Realtek to get a full answer. It looks like this controls whether the magnetics in the adapter have their center taps connected to ground (normal) or left open. The normal condition is to connect the center taps through a capacitor to ground.  Picture of what I mean: https://microchip.my.site.com/s/article/Can-I-use-magnetics-with-center-taps-joined-together-on-an-Ethernet-Switch-Design (picture is of it done wrong, normally each center tap gets its own capacitor). It is possible this is an option useful for 10/100 links that refuse to negotiate due to the taps being connected together through one capacitor. Also possible it's to work around some weird POE switches being problematic.

I did notice ESOS moved this option to debug (see their changelog: https://github.com/quantum/esos/blob/master/ChangeLog ), it's probably not an option you'd want unless testing the adapter in a lab or owning an adapter where they screwed up the hardware implementation.

closed 2023-08-22 by HerMajestyDrMona