wget/realtek-r8152-linux #28: New kernel rev new problems with r8152

preese, 2023-03-28:
I thought and hoped that the issue I have with r8152 was along the same line as with earlier kernel updates, a few mods to the compatibility.h file and things would be back on track.

I think this time, there may be deeper issues.

I use the r8152 driver code with dkms to keep things going as the kernel moves ahead.  I've done the following tests:
1. ran through the dkms steps and don't see any errors pop up.  In brief the steps are:
      dkms add -m r8152 -v 2.16.3 
      dkms build -m r8152 -v 2.16.3
      dkms install -m r8152 -v 2.16.3
2. I moved into the source dir and did 'make', shortly thereafter a r8152.ko r8152.o and some modules file were in the dir.  No errors again.

The new r8152 module does get properly added and used, 'ethtool eno2.xxx' shows the correct info for an updated driver, as does 'ethtool -i eno2.xxx'.

On one machine the USB nic completely disappears after a dkms rebuild and reboot!   I've rebuilt the host several times with similar results.  A second machine does show the USB nic name (and the ethtool info was gathered from this host).  The really weird thing about this host is that while 'ip a' shows the expected contents other than IP, looking at the nic via nmtui shows the correct nic name, eno2.xx BUT shows it without any MAC address!

These tests are on Fedora 38 beta with a 6.2.8-300.fc38.x86_64 kernel. (I first noticed this just after 6.2.x showed up)

Anyone else having these issues or know where the look?

torehl, 2023-06-03:
Installed and built it on 6.3.5 fc37. Works fine. Did you rebuilt initramfs after installing it? 

# dkms status r8152
r8152/2.16.3-20221209, 6.3.5-100.fc37.x86_64, x86_64: installed (original_module exists)

[root@srl-torel01 src]# modinfo r8152 | head
filename:       /lib/modules/6.3.5-100.fc37.x86_64/extra/r8152.ko.xz
version:        v2.16.3 (2022/07/06)
license:        GPL
description:    Realtek RTL8152/RTL8153 Based USB Ethernet Adapters
author:         Realtek nic sw <nic_swsd@realtek.com>
srcversion:     76D46D7ED32D60587B9CFD9
alias:          usb:v2BAFp0012d*dc*dsc*dp*ic02isc0Dip00in*

hcartiaux, 2024-11-14:
No activity in 1+ year, I close this issue.

If you can reproduce with the latest module version, please reopen a ticket with more info (distribution, kernel version and USB NIC model).

closed 2024-11-14 by hcartiaux