wget/realtek-r8152-linux #30: Add support for ASUS C2500

marktayl, 2023-06-10:
[ASUS USB-C2500](https://www.amazon.com/gp/product/B093FB9QWB/)

Worked for me, using 2.16.3.

```diff
--- realtek-r8152-linux/r8152.c	2023-06-08 14:36:30.267321495 -0400
+++ r8152.c	2023-06-08 12:15:49.308410187 -0400
@@ -807,6 +807,7 @@
 #define VENDOR_ID_LINKSYS		0x13b1
 #define VENDOR_ID_NVIDIA		0x0955
 #define VENDOR_ID_TPLINK		0x2357
+#define VENDOR_ID_ASUS			0x0b05
 
 #define MCU_TYPE_PLA			0x0100
 #define MCU_TYPE_USB			0x0000
@@ -20846,6 +20847,8 @@
 	/* Getac */
 	REALTEK_USB_DEVICE(0x2baf, 0x0012),
 
+	/* ASUS */
+	REALTEK_USB_DEVICE(VENDOR_ID_ASUS, 0x1976),
 	{}
 };
 
```

Creating a good `dkms.conf` for my Ubuntu 22.04.2 LTS wasn't fun, however.

jayofdoom, 2023-06-17:
I don't know what distribution you're targeting, but I'm going to propose attempting to pick this patch up in the next gentoo release of `net-misc/r8152`. Thanks for the contribution.

hcartiaux, 2023-09-03:
@delta-whiplash this is off topic, please do not spam the issues

Included in 2.7.1. 

I close this issue.

closed 2023-09-03 by hcartiaux

delta-whiplash, 2023-09-03:
No it's linked because this guy said he created a dkms.conf file