wget/realtek-r8152-linux #31: 'rtl8152_get_version' exported twice

shaperilio, 2023-06-10:
Trying to build your latest release on a Raspberry Pi 400 with kernel 6.1.21-v7l+, and I'm getting this error.

```
barf@pi400:~/Downloads/realtek-r8152-linux-2.16.3.20221209 $ make modules
make -C /lib/modules/6.1.21-v7l+/build M=/home/barf/Downloads/realtek-r8152-linux-2.16.3.20221209 modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v7l+'
  CC [M]  /home/barf/Downloads/realtek-r8152-linux-2.16.3.20221209/r8152.o
  MODPOST /home/barf/Downloads/realtek-r8152-linux-2.16.3.20221209/Module.symvers
ERROR: modpost: /home/barf/Downloads/realtek-r8152-linux-2.16.3.20221209/r8152: 'rtl8152_get_version' exported twice. Previous export was in vmlinux
make[2]: *** [scripts/Makefile.modpost:126: /home/barf/Downloads/realtek-r8152-linux-2.16.3.20221209/Module.symvers] Error 1
make[1]: *** [Makefile:1964: modpost] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v7l+'
make: *** [Makefile:24: modules] Error 2
```

TheOzzOne, 2023-08-22:
Getting the same on an RPi 4, kernel is aarch64 6.1.21-v8+ - making / building appears to be failing, but I thought it was me.

ntaken, 2023-10-15:
It is solved here https://github.com/awesometic/realtek-r8152-dkms/issues/11 .

The problem is the line "EXPORT_SYMBOL_GPL(rtl8152_get_version);" in file r8152.c.

comment it out like 

> //EXPORT_SYMBOL_GPL(rtl8152_get_version);

And you will be able to make the driver.

hcartiaux, 2024-04-05:
It's because your kernel has been compiled with rtl8152 built-in.

The solution is very probably to modify your kernel compilation config file (or using `make menuconfig`) to compile rtl8152 as a module, this way the symbol rtl8152_get_versions should not be reported in `/proc/kallsyms ` when the module is not loaded.

I close this ticket.

closed 2024-04-05 by hcartiaux