wget/realtek-r8152-linux #39: ERROR: modpost: "skb_gso_segment" [/root/doc/r8152-2.17.1/r8152.ko] undefined!

ad2248, 2024-01-31:
This can fix this error in compile.
`#include <net/gso.h>`

hcartiaux, 2024-04-05:
Looks like it's already included

```
#if defined __has_include
#if __has_include (<net/gso.h>)
#include <net/gso.h>
#endif
#endif
```

I don't know why this is not used in your environment... How can we reproduce, what's your distribution, kernel, compiler ?

ad2248, 2024-04-06:
> Looks like it's already included
> 
> ```
> #if defined __has_include
> #if __has_include (<net/gso.h>)
> #include <net/gso.h>
> #endif
> #endif
> ```
> 
> I don't know why this is not used in your environment. How can we reproduce, what's your distribution, kernel, compiler ?

Here is my system info:
distribution: debian bookworm
kernel: 6.6.15-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.6.15-2 (2024-02-04) x86_64 GNU/Linux
compiler: gcc (Debian 13.2.0-13) 13.2.0

ad2248, 2024-04-06:
I just tried compiling again and found the issue resolved. It's truly miraculous. Apologies for any inconvenience caused
 : )

closed 2024-04-06 by ad2248