wget/realtek-r8152-linux #42: incompatible-pointer-types

lexthedev, 2024-06-07:
Thanks for your work!
Facing a problem with my pc: 
  Kernel: 6.9.3-1-default arch: x86_64 bits: 64 compiler: gcc v: 13.3.0
  Desktop: KDE Plasma v: 6.0.5 Distro: openSUSE Tumbleweed 20240606

error while using make command:

make -C /lib/modules/6.9.3-1-default/build M=/home/user/Git/realtek-r8152-linux modules
make[1]: Entering directory '/usr/src/linux-6.9.3-1-obj/x86_64/default'
  CC [M]  /home/user/Git/realtek-r8152-linux/r8152.o
/home/user/Git/realtek-r8152-linux/r8152.c:19535:20: error: initialization of ‘int (*)(struct net_device *, struct ethtool_keee *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_eee *)’ [-Werror=incompatible-pointer-types]
19535 |         .get_eee = rtl_ethtool_get_eee,
      |                    ^~~~~~~~~~~~~~~~~~~
/home/user/Git/realtek-r8152-linux/r8152.c:19535:20: note: (near initialization for ‘ops.get_eee’)
/home/user/Git/realtek-r8152-linux/r8152.c:19536:20: error: initialization of ‘int (*)(struct net_device *, struct ethtool_keee *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_eee *)’ [-Werror=incompatible-pointer-types]
19536 |         .set_eee = rtl_ethtool_set_eee,
      |                    ^~~~~~~~~~~~~~~~~~~
/home/user/Git/realtek-r8152-linux/r8152.c:19536:20: note: (near initialization for ‘ops.set_eee’)
/home/user/Git/realtek-r8152-linux/r8152.c:20215:4: warning: no previous prototype for ‘rtl8152_get_version’ [-Wmissing-prototypes]
20215 | u8 rtl8152_get_version(struct usb_interface *intf)
      |    ^~~~~~~~~~~~~~~~~~~

Hope you can help or fix

jayofdoom, 2024-06-11:
I'm getting a similar error, filed https://bugs.gentoo.org/934074 about it for Gentoo tracking purposes.

jayofdoom, 2024-06-12:
#41 fixed this issue for me, and is being carried downstream in the gentoo package (well, as soon as the PR merges it will)

https://github.com/gentoo/gentoo/pull/37125

hcartiaux, 2024-07-01:
#41 merged and 2.18.1 imported, this is fixed.

closed 2024-07-01 by hcartiaux