wget/realtek-r8152-linux #50: Realtek 5 GbE USB Network Dongles (WisdPi, but generic issue with 8157) cause NFS hangs and kernel crashes

jerryabramson, 2025-07-04:
About 8 months ago I started playing around with the 8157 USB 5 GbE dongles from WisdPi and Proxmox.

A few other people have tried these dongles on Proxmox, with forum posts on both proxmox.com and servethehome.com

The first requirement was to be on Kernel 6.11 or newer, and to compile the source code from Realtek.

At the time I tried the 2.19.2 drivers that were released last fall.

For typical Networking issues, I had no real issues, and saw about 4.7 Gbs upload/downloads, and there were no stability issues.

I then tried to move over my Proxmox server that uses NFS to host many of the QCOW2 files for my cluster.

- Note I had been hosting these using 2.5 GbE USB dongles for several years without any issues.

I then move over my cluster to 5 GbE dongles and a quality Microtik 10 GbE switch.

With the 5 GbE dongles, and the driver built from Realtek source (2.19.2) the systems then became incredibly unstable.

There were many kernel hangs, both on the nodes acting as NFS clients, as well as the node acting as an NFS server.

Many QCOW2 files became corrupted (I had plenty of backups, so no loss except for downtime).

I am hesitant to try out either the 2.20.1 driver from Realtek, or either your driver, or the forked wget driver.

There was a hint in one of the forum posts upload NFS fragmentation, and hinted it might be an NFS issue, and not related to the 5 GbE dongles.

I am suspicious of this theory, since I moved over the cluster to OWC 10 GbE thunderbolt dongles, and the system has been very stable for 9+ months.

It definitely has something to do with an NFS interaction with the Realtek drivers.

Any ideas??

- https://forums.servethehome.com/index.php?threads/realtek-rtl8157-5-gbe-adapters-with-10-gb-usb-interface-wavlink-wisdpi.46098/
- 
- https://forums.servethehome.com/index.php?threads/realtek-rtl8157-5-gbe-adapters-with-10-gb-usb-interface-wavlink-wisdpi.46098/post-467917
- 
- https://forum.proxmox.com/threads/opt-in-linux-6-11-kernel-for-proxmox-ve-8-available-on-test-no-subscription.156818/post-780133
-

popic74, 2025-11-28:
I'm also testing the new version of the 2.21.4 driver. It works great. No more kernel crashes on Debian 12 with kernel 6.12.

luckman212, 2026-10-23:
@sgueko wondering if your systems have continued to been stable? Have you tried the 2.22.1 driver? I also have PVE 9 + NFS + iSCSI + rtl8157 and ran into #65 yesterday. So trying to narrow it down.

In both the 2.21.4 and the 2.22.1 drivers, `RTL8152_MAX_RX` is already declared at
```
#define RTL8152_MAX_RX		10
```

and `RTL_MAX_SG_NUM` is defined as 

```
#define RTL_MAX_SG_NUM		(MAX_SKB_FRAGS + 1)
```
Are you still using those modified values? My system has been pretty stable with 2.21.4 but after installing 2.22.1 it crashed fairly quickly. Could be one of the new kernels too - the 7.x kernels have been coming out 1-2 per week with the flood of new AI generated CVEs…

jerryabramson, 2026-10-23:
For now I have stayed with the Thunderbolt 10 GbE dongles.

They are fast, and hot! I have them on top of USB fans.

After losing my cluster a few times with the Realtek dongles I am only using the Realtek’s on standard clients (laptops, etc…). 

It’s a shame that ancient Acquantis chips survived this long.

Once the 7.2 kernel has been out for a few months and Proxmox has endorsed them, would I consider going back to the Realtek’s.

I have kept up with the code changes being made to the Realtek drivers in 7.2 and 7.3 and they are significant.

Nvidia and AMD also did a bunch of work on the drivers, since their development boxes use integrated Realtek Ethernet ports. I don’t know if all their changes have made it back into the mainline kernel.